bigbugcc/OpenWrts #32: 保存并应用失效

adfnkjasd, 2023-10-31:
更改任何设置后点击保存并应用都会失效
[image]

adfnkjasd, 2023-10-31:
追加bug，会自己配置虚拟网口和负载平衡
[image]

bigbugcc, 2023-11-01:
> 更改任何设置后点击保存并应用都保存不上，右上角还是有未保存配置，点进去保存也没用
> 
> [image]
> 
> 

可以先谷歌一下 很多解决方案，并非固件问题

closed 2023-11-11 by bigbugcc

scolia, 2023-11-26:
不知道这个回复算不算晚，这是上游lede固件的问题，在我的树莓派3b中得到了复现，其表现表现为reboot之后，系统的根路径会被挂载为只读，因此所有的设置都无法被保存（因为只读不可写入）。其最简单的解决方案，ssh进去，手动挂着根分区为rw状态（具体操作百度），然后趁机在openwrt界面中选择：系统-挂载点，拖到下面，把/rom挂载前的勾给去掉，然后重启，应该以后就会恢复成读写状态。

其根本原因，好像是lede固件魔改 fstool 工具，截止到我回复的时间，此bug仍然未修复……

adfnkjasd, 2023-11-26:
> 不知道这个回复算不算晚，这是上游固件的问题，在我的树莓派3b中得到了复现，其表现表现为reboot之后，系统的根路径会被挂载为只读，因此所有的设置都无法被保存（因为只读不可写入）。
> 
> [coolsnowwolf/lede#11662](https://github.com/coolsnowwolf/lede/pull/11662) [coolsnowwolf/lede#10317](https://github.com/coolsnowwolf/lede/issues/10317)
> 
> 看了下，官方目前还没有修复，此bug应该只针对于ext4文件系统。
> 
> 最简单的解决方案是ssh进去，手动挂着根分区为rw状态（具体操作百度），然后趁机在openwrt界面中选择：系统-挂载点，拖到下面，把/rom挂载前的勾给去掉，保存。之后重启应该就是读写状态了。
> 
> 当然，如果你现在处于可读写状态（即没有进行第一次reboot），那么立即这样操作就行，这样就不用ssh进去挂载了，省点功夫。

确实，当时提完issue就解决了，忘了关，其实就是重启后硬盘变为只读，也很好解决，进ssh或者终端，写个开机重新挂载就行了