bigbugcc/OpenWrts #44: Build failure: `System.IO.IOException: No space left on device`

jayagami, 2024-07-04:
Building failed during the Compile step.

```

System.IO.IOException: No space left on device : '/home/runner/runners/2.317.0/_diag/Worker_20240703-031301-utc.log'

```

bigbugcc, 2024-07-04:
just only `No space left on device`, you could free space to build.

jayagami, 2024-07-04:
> just only `No space left on device`, you could free space to build.

Thanks for replying.
I was building with GitHub Actions, so GitHub Actions is not available for building this project?
I noticed there is already a free step in the action steps.
I am trying to build it with a self-hosted action runner now.

bigbugcc, 2024-07-04:
If you just want to compile your own firmware, please fork this project and modify the configurations in the `configs` directory and trigger the corresponding `action`. Do not modify the configuration files in workflow folder without understanding the `action` configuration parameters, which can cause problems with your compilation process.

jayagami, 2024-07-06:
> If you just want to compile your own firmware, please fork this project and modify the configurations in the `configs` directory and trigger the corresponding `action`. Do not modify the [...]

Thanks for your support. It's weird, but I re-forked this repo and successfully recompiled the firmware.

closed 2024-07-06 by jayagami