bigbugcc/OpenWrts #45: 树莓派5 ipv6

xlccccc, 2024-07-07:
我用树莓派5刷入该openwrt固件
光猫为路由模式，ip是 192.168.1.1
openwrt的lan口配置为 ip 192.168.1.10 网关 192.168.1.1，开启DHCP服务器以及DHCPV6（只配置了一个lan口）
手机电脑连接openwrt且将192.168.1.10作为网关可以分配到公网ipv6
但作为openwrt的树莓派无公网ipv6，应该如何配置呢
[image]

bigbugcc, 2024-07-07:
旁路网关获取ipv6你可以试着添加一个新的接口，接口协议改成`DHCPv6客户端`,并划分防火墙为`lan域`。

xlccccc, 2024-07-07:
树莓派五的固件好像没有`DHCPv6客户端`，是我下的有问题m吗
[image]
我尝试过lan口改为`192.168.10.1`（其他段），然后设置dhcp服务器，接口用无线网卡
再新建一个wan口作为dhcp客户端，不设置服务器，但是此时自动获取的网关还是光猫的192.168.1.1，将网关换到192.168.10.1是连不上网的，且wan口一直没获取到ip，是什么问题呢
[image]

xlccccc, 2024-07-07:
我固定ipv6是可以用公网ipv6了，但是还是想不通为什么dhcp获取不到。。

bigbugcc, 2024-07-07:
> 树莓派五的固件好像没有`DHCPv6客户端`，是我下的有问题m吗 [...]

1. 固件的`DHCPv6`客户端，我会尽快排查解决。
2. 后续问题的描述不太清楚，通常情况下如果你打算创建一个wan口得有对应的物理设备(可以是桥接)，且连接对应的上游网络才行，其次防火墙的域也要注意(没有网络)。DHCP未正常工作设备肯定获取不到ip，按你之前的描述(旁路网关)，你的路由大致应该关闭了DHCP 转由网关接手DHCP了，在接入WAN也就获取不到了，可以多看网上的旁路网关的教程就能搞明白。

bigbugcc, 2024-07-07:
> 我固定ipv6是可以用公网ipv6了，但是还是想不通为什么dhcp获取不到。。

通常一个设备域不能即做LAN又做WAN，LAN域ipv4的DHCP服务器就是这个道理。

xlccccc, 2024-07-07:
嗯嗯 麻烦师傅了 还有就是现在固件默认的源有的404了，是不是需要改改
[image]

xlccccc, 2024-07-07:
[image]
然后装的python3是会报错的

bigbugcc, 2024-07-07:
> 嗯嗯 麻烦师傅了 还有就是现在固件默认的源有的404了，是不是需要改改 [...]

固件自带的软件源，可自行替换阿里、腾讯、中科大。

closed 2024-07-12 by bigbugcc