bigbugcc/OpenWrts #50: Raspberry PI 5 did not compile

bxiao0719, 2024-09-30:
Raspberry PI 5 did not compile

bxiao0719, 2024-09-30:
Applying /workdir/openwrt/target/linux/bcm27xx/patches-6.1/950-0078-Revert-Bluetooth-Always-request-for-user-confirmatio.patch using plaintext: 
patching file net/bluetooth/smp.c
Hunk #1 FAILED at 2214.
Hunk #2 succeeded at 2223 (offset -6 lines).
1 out of 2 hunks FAILED -- saving rejects to file net/bluetooth/smp.c.rej
Patch failed!  Please fix /workdir/openwrt/target/linux/bcm27xx/patches-6.1/950-0078-Revert-Bluetooth-Always-request-for-user-confirmatio.patch!
make[4]: *** [Makefile:43: /workdir/openwrt/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/linux-6.1.111/.prepared_aff15fc8e919fa2c02eeadbfa49e5cad] Error 1
make[4]: Leaving directory '/workdir/openwrt/target/linux/bcm27xx'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#9.61#8.41#13.25
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_cortex-a76_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

bigbugcc, 2024-10-01:
Firmware compilation fixed.

closed 2024-10-01 by bigbugcc