bigbugcc/OpenWrts #52: 所以之前的版本有没有留档，新的JS的luci太难用了，而且很多插件都不支持了

adfnkjasd, 2024-10-23:
### 详细叙述

lede更了新的luci，很多之前的插件都用不了了，docker更是直接瘫痪

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

openwrt

### 详细日志

10月8日之前的

bigbugcc, 2024-10-24:
确实如此，luci拉取的还是[18.版本](https://github.com/bigbugcc/OpenWrts/blob/399a4907203c2861ed3a3c649bec2789d75cc35e/configure.sh#L22)，我目前使用的还是两个月前编译的固件😅。大致看了下仓库 大雕在17号修改了源，这个更新挺大的，我暂时修改源至老版本等待其它插件更新同步。

adfnkjasd, 2024-10-24:
> 确实如此，luci拉取的还是[18.版本](https://github.com/bigbugcc/OpenWrts/blob/399a4907203c2861ed3a3c649bec2789d75cc35e/configure.sh#L22)，我目前使用的还是两个月前编译的固件😅。大致看了下仓库 大雕在17号修改了源，这个更新挺大的，我暂时修改源至老版本等待其它插件更新同步。

请问还有老版本的固件吗，如果有的话老版本固件放一下Releases吧，现在这个怎么用怎么难受😅，感谢

bigbugcc, 2024-10-24:
老版本的固件没有留存😂，Action正在重新编译了，大概2-3个小时，稍等下。

adfnkjasd, 2024-10-24:
> 老版本的固件没有留存😂，Action正在重新编译了，大概2-3个小时，稍等下。

ok，那我这里就close了

closed 2024-10-24 by adfnkjasd