bigbugcc/OpenWrts #53: 有关固件简化的讨论

bigbugcc, 2024-10-25:
征求大家对`iStore`软件商店的看法和需求
- iStore商店目前支持x86和arm设备   

📕iStore的加入确实会给固件包减负，稳定的同时也有很高的可玩性(取决于自己)，固件本身只保留基础的app，其它均自行到软件商店添加。但是也会带来一些问题，比如固件升级需手动备份iStore配置，其次有些app也可能无法安装或出错。

🎉x86_Lite固件包含了`iStore`，经过几个月的使用个人感觉需求并不大，主要使用场景也就magic插件、dns优化和广告过滤。

大家可以聊一下自己的需求及看法....

adfnkjasd, 2024-10-25:
可有可无吧，个人感觉软路由系统最重要是稳定可用，不是可玩性高，要可玩高可以玩docker之类的，openwrt这种还是稳定的当一个梯子加猫的功能就好。（顺便问下最新的action的luci用回老的18了吗，新的23适配真的依托答辩）

bigbugcc, 2024-10-25:
> 可有可无吧，个人感觉软路由系统最重要是稳定可用，不是可玩性高，要可玩高可以玩docker之类的，openwrt这种还是稳定的当一个梯子加猫的功能就好。（顺便问下最新的action的luci用回老的18了吗，新的23适配真的依托答辩）

还是稳定刚需🙋‍♂️，Lite版固件移除大部分插件后保留docker也是这个目的(稳定的同时，也留足了可玩性)，而且几乎不会影响系统的稳定性。
- 最新的固件已经回退18了。

wention, 2024-10-25:
需要讨论的内容是维护一份核心插件列表版本（其他新颖插件可以酌情考虑出个技术预览版）

bigbugcc, 2024-10-26:
> 需要讨论的内容是维护一份核心插件列表版本（其他新颖插件可以酌情考虑出个技术预览版）

目前[Lite版的插件](https://github.com/bigbugcc/OpenWrts/blob/main/configs/LuciApp_Lite.config)就仅保留必要的，其它都通过iStore和docker自己折腾。以Lite版本的配置看下有什么要补充的插件。

adam7bf, 2025-02-15:
比较赞同稳定优先，至于是否精简可以从另一个角度考虑，比如我每次会手动卸载我不使用的程序包，因此预置app丰富一些对我没太大影响。不过最近自动编译的x86版本里ssr是不是有点问题，服务可以开启，但是一访问服务就自动重启~~~

closed 2025-03-30 by bigbugcc