bigbugcc/OpenWrts #54: 没有规律的重启

adfnkjasd, 2024-11-07:
### 问题描述

查了各种原因，原来以为是turboacc里的shortcutfe导致的，结果关了以后还是一样会不规律重启，有时候一天一次有时候一天多次

### 重复 issue

- [X] 没有类似的 issue

### 浏览器搜索

- [X] 未找到类似的问题

### 设备型号

x86_64

### 固件版本

openwrt-x86-64-generic-ext4-combined-efi.img.gz

### 详细日志

过长

adfnkjasd, 2024-11-07:
日志
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.uptime/main.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/system.cpu/irq.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_space._tmp/used.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/system.cpu/user.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.uptime/uptime.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/system.cpu/system.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_space._tmp/reserved_for_root.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.clients/main.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/system.cpu/nice.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/system.cpu/iowait.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/system.cpu/idle.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.clients/clients.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_inodes._tmp/main.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_inodes._tmp/avail.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.requests/main.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.requests/requests.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/cpu.cpu0/main.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_inodes._tmp/used.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/cpu.cpu0/guest_nice.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.net/main.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_inodes._tmp/reserved_for_root.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.net/in.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/cpu.cpu0/guest.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_space._/main.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.net/out.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/cpu.cpu0/steal.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_space._/avail.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.response_time/main.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/cpu.cpu0/softirq.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/netdata.response_time/average.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_space._/used.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/cpu.cpu0/irq.db.
Thu Nov  7 07:34:00 2024 daemon.info netdata[11340]: Initializing file /var/cache/netdata/disk_space._/reserved_for_root.db.
Thu Nov  7 07:34:01 2024 daemon.info pppd[10960]: syncppp not active
Thu Nov  7 07:34:01 2024 daemon.info pppd[10960]: Remote message: 5;User(053102729706) Checking LM, (online count=1)>=(Max num=1)., Request Deny by 110
Thu Nov  7 07:34:01 2024 daemon.err pppd[10960]: PAP authentication failed
Thu Nov  7 07:34:01 2024 daemon.notice pppd[10960]: Modem hangup
Thu Nov  7 07:34:01 2024 daemon.notice pppd[10960]: Connection terminated.
Thu Nov  7 07:34:01 2024 daemon.info pppd[10960]: Exit.
Thu Nov  7 07:34:01 2024 daemon.notice netifd: Interface 'wan' is now down
Thu Nov  7 07:34:01 2024 daemon.info avahi-daemon[4104]: Interface eth0.IPv6 no longer relevant for mDNS.
Thu Nov  7 07:34:01 2024 daemon.info avahi-daemon[4104]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::231:11ff:fe06:1b0d.
Thu Nov  7 07:34:01 2024 daemon.info avahi-daemon[4104]: Withdrawing address record for fe80::231:11ff:fe06:1b0d on eth0.
Thu Nov  7 07:34:01 2024 daemon.notice netifd: Interface 'wan' is disabled
Thu Nov  7 07:34:01 2024 daemon.notice netifd: Interface 'wan' is enabled
Thu Nov  7 07:34:01 2024 daemon.notice netifd: Interface 'wan' is setting up now
Thu Nov  7 07:34:01 2024 daemon.notice netifd: Network device 'eth0' link is down
Thu Nov  7 07:34:01 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Thu Nov  7 07:34:01 2024 kern.info kernel: [   32.894489] 8021q: adding VLAN 0 to HW filter on device eth0
Thu Nov  7 07:34:01 2024 daemon.info pppd[12559]: Plugin pppoe.so loaded.
Thu Nov  7 07:34:01 2024 daemon.info pppd[12559]: PPPoE plugin from pppd 2.4.9
Thu Nov  7 07:34:01 2024 daemon.notice pppd[12559]: pppd 2.4.9 started by root, uid 0
[... cut ...]
Thu Nov  7 15:34:12 2024 daemon.info netdata[11340]: RRDSET: chart name 'netdata.plugin_chartsd_sensors' on host 'OpenWrt' already exists.
Thu Nov  7 19:03:12 2024 daemon.err uhttpd[3919]: sh: out of range
Thu Nov  7 19:03:12 2024 daemon.err uhttpd[3919]: sh: out of range
Thu Nov  7 19:06:59 2024 daemon.err uhttpd[3919]: sh: out of range
Thu Nov  7 19:07:00 2024 daemon.err uhttpd[3919]: sh: out of range
Thu Nov  7 19:07:05 2024 daemon.err uhttpd[3919]: sh: out of range
Thu Nov  7 19:07:10 2024 daemon.err uhttpd[3919]: sh: out of range
Thu Nov  7 19:07:15 2024 daemon.err uhttpd[3919]: sh: out of range

bigbugcc, 2024-11-08:
没检索到关键错误信息，wan口一直在重启 把mentohust关了试试。

adfnkjasd, 2024-11-08:
> 没检索到关键错误信息，wan口一直在重启 把mentohust关了试试。

mentohust重试3次以后就不再重试了，而且这是更新过一次的固件，上一版固件直接把mentohust删除了，还是会重启

adfnkjasd, 2024-11-08:
> 没检索到关键错误信息，wan口一直在重启 把mentohust关了试试。

已经把日志保存到外部存储了，只能等下次无规律重启以后再看看有没有什么报错了

bigbugcc, 2024-11-08:
内核日志检索一下看看

adfnkjasd, 2024-11-09:
> 内核日志检索一下看看

没有什么有用的，我再升级一下看看吧

adfnkjasd, 2024-11-09:
> 内核日志检索一下看看

抓到上一次重启的日志，但最后的时间是10点，重启的时间是13点。现在有点手足无措

Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: ** 失败等待:	15秒
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: ** 允许失败:	0次
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: ** 组播地址:	标准
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: ** DHCP方式:	不使用
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: !! 在网卡dummy0上获取IP失败!
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: !! 在网卡dummy0上获取子网掩码失败!
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: ** 本机MAC:	7a:40:8b:74:e8:ed
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: ** 使用IP:	0.0.0.0
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: ** 子网掩码:	255.255.255.255
Sat Nov  9 10:03:04 2024 kern.info kernel: [   62.572321] dummy0: entered promiscuous mode
Sat Nov  9 10:03:04 2024 kern.info kernel: [   62.573053] dummy0: left promiscuous mode
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: !! 打开网卡dummy0失败: dummy0: That device is not up
Sat Nov  9 10:03:04 2024 daemon.info mentohust[20050]: >> 认证已退出。
Sat Nov  9 10:03:05 2024 daemon.err uhttpd[3276]: sh: /etc/init.d/dockerman: line 6: config_load: not found
Sat Nov  9 10:03:05 2024 daemon.err uhttpd[3276]: sh: /etc/init.d/dockerman: line 130: extra_command: not found
Sat Nov  9 10:03:05 2024 daemon.err uhttpd[3276]: sh: /etc/init.d/dockerman: line 131: extra_command: not found
Sat Nov  9 10:03:06 2024 daemon.err uhttpd[3276]: env: can't execute '/etc/init.d/timewol': Permission denied
Sat Nov  9 10:03:06 2024 kern.err kernel: [   64.945177] squashfs: Unknown parameter 'discard'
Sat Nov  9 10:03:06 2024 kern.info kernel: [   64.945597] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x89e45589)
Sat Nov  9 10:03:06 2024 kern.err kernel: [   64.945778] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
Sat Nov  9 10:03:06 2024 kern.info kernel: [   64.946426] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x14a1a27d)
Sat Nov  9 10:03:06 2024 kern.err kernel: [   64.946608] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
Sat Nov  9 10:03:06 2024 kern.err kernel: [   64.949689] exFAT-fs (sda128): invalid boot record signature
Sat Nov  9 10:03:06 2024 kern.err kernel: [   64.949837] exFAT-fs (sda128): failed to read boot sector
Sat Nov  9 10:03:06 2024 kern.err kernel: [   64.949977] exFAT-fs (sda128): failed to recognize exfat type
Sat Nov  9 10:03:06 2024 kern.err kernel: [   64.957062] ntfs3: sda128: Primary boot signature is not NTFS.
Sat Nov  9 10:03:06 2024 kern.err kernel: [   64.957242] ntfs3: sda128: try to read out of volume at offset 0x3ba00
Sat Nov  9 10:03:09 2024 daemon.info frpc[21450]: 2024/11/09 02:03:09 [1;34m[I] [root.go:220] start frpc service for config file [/var/etc/frpc.ini][0m
Sat Nov  9 10:03:09 2024 daemon.info frpc[21450]: 2024/11/09 02:03:09 [1;33m[W] [service.go:133] login to server failed: dial tcp 127.0.0.1:7000: connect: connection refused[0m
Sat Nov  9 10:03:09 2024 daemon.info frpc[21450]: 2024/11/09 02:03:09 [1;34m[I] [root.go:236] frpc service for config file [/var/etc/frpc.ini] stopped[0m
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]:
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: 欢迎使用MentoHUST	版本: 0.3.1
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: Copyright (C) 2009-2010 HustMoon Studio
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: 人到华中大，有甜亦有辣。明德厚学地，求是创新家。
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: Bug report to http://code.google.com/p/mentohust/issues/list
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]:
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: !! 打开libnotify失败，请检查是否已安装该库文件。
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 网卡[1]:	eth0
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 网卡[2]:	eth1
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 网卡[3]:	br-lan
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 网卡[4]:	pppoe-wan
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 网卡[7]:	docker0
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 网卡[8]:	sit0
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 网卡[9]:	dummy0
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ?? 请选择网卡[1-9]: ** 您选择了第[9]块网卡。
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ?? 请输入用户名: ?? 请输入密码: ?? 请选择组播地址(0标准 1锐捷私有 2赛尔): ?? 请选择DHCP方式(0不使用 1二次认证 2认证后 3认证前): ** 用户名:
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 网卡: 	dummy0
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 认证超时:	8秒
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 响应间隔:	30秒
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 失败等待:	15秒
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 允许失败:	0次
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 组播地址:	标准
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** DHCP方式:	不使用
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: !! 在网卡dummy0上获取IP失败!
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: !! 在网卡dummy0上获取子网掩码失败!
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 本机MAC:	7a:40:8b:74:e8:ed
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 使用IP:	0.0.0.0
Sat Nov  9 10:03:09 2024 daemon.info mentohust[21468]: ** 子网掩码:	255.255.255.255
Sat Nov  9 10:03:09 2024 kern.info kernel: [   67.644154] dummy0: entered promiscuous mode
Sat Nov  9 10:03:09 2024 kern.info kernel: [   67.644887] dummy0: left promiscuous mode
[... cut ...]
Sat Nov  9 10:06:36 2024 authpriv.info dropbear[7353]: Early exit: Terminated by signal
Sat Nov  9 10:06:36 2024 authpriv.info dropbear[12304]: Early exit: Terminated by signal
Sat Nov  9 10:06:36 2024 daemon.notice ttyd[16375]: [2024/11/09 10:06:36:7062] N: received signal: SIGTERM (15), exiting...
Sat Nov  9 10:06:36 2024 daemon.notice ttyd[16375]: [2024/11/09 10:06:36:7062] N: send ^C to force exit.
Sat Nov  9 10:06:36 2024 daemon.notice rpc.mountd[15190]: Caught signal 15, un-registering and exiting.

adfnkjasd, 2024-11-09:
哦对，描述不太准，更像是断电后上电自动启动，因为如果是重启的话宽带拨号不会等待5分钟的重新下发ip，目前的情况是重启后要等待宽带拨号ip重新下发的5分钟，所以现在应该是没有规律的断电后重新上电启动

bigbugcc, 2024-11-11:
日志里的报错也不足以引起系统重启，这让我想到之前`j4125`的机器，也是会莫名奇妙的重启🤣；或者你试一下看纯净版的固件，看是不是某些插件引起的。我目前用的是x86 Lite的固件没发现有什么问题。

adfnkjasd, 2024-11-11:
> 日志里的报错也不足以引起系统重启，这让我想到之前`j4125`的机器，也是会莫名奇妙的重启🤣；或者你试一下看纯净版的固件，看是不是某些插件引起的。我目前用的是x86 Lite的固件没发现有什么问题。

lite应该是没有turbo acc的插件吧，我把turboacc删了之后现在2天了还没出现断电重启的现象，之前只是关了shortcut FE。

bigbugcc, 2024-11-11:
Lite版本也有，[turboacc](https://github.com/bigbugcc/OpenWrts/blob/53e11b6bc841cfed63b2439e0e36a4a009e039a3/configs/luci/Lite.config#L46C1-L48C52) 自己fork一份编译把`y`改成`n`，你用的啥机器😂

adfnkjasd, 2024-11-11:
> Lite版本也有，[turboacc](https://github.com/bigbugcc/OpenWrts/blob/53e11b6bc841cfed63b2439e0e36a4a009e039a3/configs/luci/Lite.config#L46C1-L48C52) 自己fork一份编译把`y`改成`n`，你用的啥机器😂

自组的，i5 4590t，现在删了turboacc以后暂时稳定了，再等几天如果没有断电重启大概就是这个的问题了，其实仔细一想x86开了也等于白开🤣，最多就是bbr可能还有点用

bigbugcc, 2024-11-11:
x86下硬件性能过剩确实作用不大，只是简单的上网需求这硬件也足够了

adfnkjasd, 2024-11-11:
> x86下硬件性能过剩确实作用不大，只是简单的上网需求这硬件也足够了

确实，那就先close了，如果出问题再reopen吧

closed 2024-11-11 by adfnkjasd