bigbugcc/OpenWrts #56: The wireless menu option is missing from the LuCI interface (AX210)

jayagami, 2025-01-10:
### 问题描述

First of all, I would like to sincerely thank the author for their hard work in creating one of the few excellent firmwares available.

However, after I updated the firmware, although lspci shows that the Intel AX210 has been loaded correctly, there is no wireless option in the web interface.

I hope wireless network support can be added. I don’t have more network cards available for testing, but as far as I know, the R5C supports this type of wireless network card.

### 重复 issue

- [X] 没有类似的 issue

### 浏览器搜索

- [X] 未找到类似的问题

### 设备型号

NanoPi R5C

### 固件版本

openwrt-rockchip-armv8-friendlyarm_nanopi-r5c-squashfs-sysupgrade.img.gz

### 详细日志

Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.069447] thermal_sys: Registered thermal governor 'step_wise'
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.069459] thermal_sys: Registered thermal governor 'power_allocator'
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.070115] cpuidle: using governor menu
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.071333] hw-breakpoint: found 6 breakpoint and 4 watchpoint registers.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.072145] ASID allocator initialised with 32768 entries
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.073405] Serial: AMBA PL011 UART driver
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.086113] /vop@fe040000: Fixed dependency cycle(s) with /hdmi@fe0a0000
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.086868] /hdmi@fe0a0000: Fixed dependency cycle(s) with /vop@fe040000
Thu Jan  9 22:05:58 2025 kern.warn kernel: [    0.100890] gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.102124] rockchip-gpio fdd60000.gpio: probed /pinctrl/gpio@fdd60000
Thu Jan  9 22:05:58 2025 kern.warn kernel: [    0.103177] gpio gpiochip1: Static allocation of GPIO base is deprecated, use dynamic allocation.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.104273] rockchip-gpio fe740000.gpio: probed /pinctrl/gpio@fe740000
Thu Jan  9 22:05:58 2025 kern.warn kernel: [    0.105279] gpio gpiochip2: Static allocation of GPIO base is deprecated, use dynamic allocation.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.106426] rockchip-gpio fe750000.gpio: probed /pinctrl/gpio@fe750000
Thu Jan  9 22:05:58 2025 kern.warn kernel: [    0.107530] gpio gpiochip3: Static allocation of GPIO base is deprecated, use dynamic allocation.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.108621] rockchip-gpio fe760000.gpio: probed /pinctrl/gpio@fe760000
Thu Jan  9 22:05:58 2025 kern.warn kernel: [    0.109622] gpio gpiochip4: Static allocation of GPIO base is deprecated, use dynamic allocation.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.110739] rockchip-gpio fe770000.gpio: probed /pinctrl/gpio@fe770000
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.115499] /hdmi@fe0a0000: Fixed dependency cycle(s) with /hdmi-con
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.116217] /hdmi-con: Fixed dependency cycle(s) with /hdmi@fe0a0000
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.121536] Modules: 2G module region forced by RANDOMIZE_MODULE_REGION_FULL
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.122247] Modules: 0 pages in range for non-PLT usage
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.122253] Modules: 519712 pages in range for PLT usage
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.124102] HugeTLB: registered 1.00 GiB page size, pre-allocated 0 pages
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.125291] HugeTLB: 0 KiB vmemmap can be freed for a 1.00 GiB page
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.125908] HugeTLB: registered 32.0 MiB page size, pre-allocated 0 pages
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.126570] HugeTLB: 0 KiB vmemmap can be freed for a 32.0 MiB page
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.127220] HugeTLB: registered 2.00 MiB page size, pre-allocated 0 pages
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.127885] HugeTLB: 0 KiB vmemmap can be freed for a 2.00 MiB page
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.128500] HugeTLB: registered 64.0 KiB page size, pre-allocated 0 pages
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.129163] HugeTLB: 0 KiB vmemmap can be freed for a 64.0 KiB page
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.132446] cryptd: max_cpu_qlen set to 1000
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.136799] iommu: Default domain type: Translated
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.137285] iommu: DMA domain TLB invalidation policy: strict mode
Thu Jan  9 22:05:58 2025 kern.notice kernel: [    0.138241] SCSI subsystem initialized
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.139515] usbcore: registered new interface driver usbfs
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.140175] usbcore: registered new interface driver hub
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.140835] usbcore: registered new device driver usb
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.142143] scmi_core: SCMI protocol bus registered
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.144260] clocksource: Switched to clocksource arch_sys_counter
Thu Jan  9 22:05:58 2025 kern.notice kernel: [    0.145357] VFS: Disk quotas dquot_6.6.0
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.145788] VFS: Dquot-cache hash table entries: 512 (order 0, 4096 bytes)
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.153218] NET: Registered PF_INET protocol family
Thu Jan  9 22:05:58 2025 kern.info kernel: [    0.154032] IP idents hash table entries: 65536 (order: 7, 524288 bytes, linear)
[... 197 lines cut ...]
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.393338] vcca1v8_image: Bringing 600000uV into 1800000-1800000uV
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.411705] rk808-rtc rk808-rtc.5.auto: registered as rtc0
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.414874] rk808-rtc rk808-rtc.5.auto: setting system clock to 2017-08-05T09:00:06 UTC (1501923606)
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.424771] sdhci: Secure Digital Host Controller Interface driver
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.425335] sdhci: Copyright(c) Pierre Ossman
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.425722] Synopsys Designware Multimedia Card Interface Driver
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.426858] sdhci-pltfm: SDHCI platform and OF driver helper
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.427822] dwmmc_rockchip fe2b0000.mmc: IDMAC supports 32-bit address mode.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.428523] dwmmc_rockchip fe2b0000.mmc: Using internal DMA controller.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.429120] dwmmc_rockchip fe2b0000.mmc: Version ID is 270a
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.429675] dwmmc_rockchip fe2b0000.mmc: DW MMC controller at irq 100,32 bit host data width,256 deep fifo
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.430834] ledtrig-cpu: registered to indicate activity on CPUs
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.432160] scmi_protocol scmi_dev.1: Enabled polling mode TX channel - prot_id:16
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.433200] arm-scmi firmware:scmi: SCMI Notifications - Core Enabled.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.433842] arm-scmi firmware:scmi: SCMI Protocol v2.0 'rockchip:' Firmware version 0x0
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.437106] usbcore: registered new interface driver usbhid
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.437608] usbhid: USB HID core driver
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.440156] hw perfevents: enabled with armv8_cortex_a55 PMU driver, 7 counters available
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.443042] NET: Registered PF_INET6 protocol family
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.444846] Segment Routing with IPv6
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.444856] mmc_host mmc0: Bus speed (slot 0) = 375000Hz (slot req 400000Hz, actual 375000HZ div = 0)
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.445211] In-situ OAM (IOAM) with IPv6
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.446418] NET: Registered PF_PACKET protocol family
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.446905] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
Thu Jan  9 22:05:58 2025 kern.info kernel: [    1.448041] 8021q: 802.1Q VLAN Support v1.8

bigbugcc, 2025-01-10:
The interface type of the wireless module (m.2 or usb ?) is required.

jayagami, 2025-01-11:
> The interface type of the wireless module (m.2 or usb ?) is required.

Hi, I have an Intel AX210 with a mini PCIe port. I ran the lspci command, and the AX210 is detected as a "Network Controller." However, there is no wireless option appearing in the network settings. I tried recompiling and adding the AX210 driver, but it didn’t have any effect.

bigbugcc, 2025-01-11:
As far as I know R5C has a M.2 interface to support Pcie2.1x1, need to add the device driver in `Kernel modules > Network Devices > iwlwifi-firmware-ax210`, I don't have that hardware device, so I can't test it for that😂.

closed 2025-01-21 by bigbugcc