bigbugcc/OpenWrts #60: Unable to open dashboards under OpenClash (YACD, METACUBEXD, ZASHBOARD, DASHBOARD)

galendai, 2025-05-15:
### 问题描述

从新跟新固件后可以正常运行，但运行一段时间之后会报404 Page not found错误

### 重复 issue

- [x] 没有类似的 issue

### 浏览器搜索

- [x] 未找到类似的问题

### 设备型号

RaspberryPi4

### 固件版本

openwrt-bcm27xx-bcm2711-rpi-4-squashfs-sysupgrade.img

### 详细日志

看起来没有有用的信息：

```
Thu May 15 00:01:17 2025 cron.err crond[11913]: crond (busybox 1.36.1) started, log level 5
Thu May 15 01:00:49 2025 cron.err crond[26045]: crond (busybox 1.36.1) started, log level 5
Thu May 15 01:14:14 2025 daemon.err netdata[9064]: read failed: end of file
Thu May 15 01:14:14 2025 daemon.info netdata[9064]: PARSER ended
Thu May 15 01:14:14 2025 daemon.err netdata[9064]: '/usr/lib/netdata/plugins.d/charts.d.plugin' (pid 4162) disconnected after 14322 successful data collections (ENDs).
Thu May 15 01:14:16 2025 daemon.info netdata[9064]: connected to '/usr/lib/netdata/plugins.d/charts.d.plugin' running on pid 29728
Thu May 15 01:14:17 2025 daemon.info netdata[9064]: RRDSET: chart name 'sensors.temp_thermal_zone0_thermal_thermal_zone0' on host 'LEDE' already exists.
Thu May 15 01:14:17 2025 daemon.info netdata[9064]: RRDSET: chart name 'netdata.plugin_chartsd_sensors' on host 'LEDE' already exists.
Thu May 15 02:00:52 2025 daemon.info netdata[9064]: Initializing file /var/cache/netdata/ipv6.sockstat6_raw_sockets/main.db.
Thu May 15 02:00:52 2025 daemon.info netdata[9064]: Initializing file /var/cache/netdata/ipv6.sockstat6_raw_sockets/inuse.db.
Thu May 15 02:00:54 2025 cron.err crond[7939]: crond (busybox 1.36.1) started, log level 5
Thu May 15 03:00:00 2025 cron.err crond[7939]: USER root pid 20967 cmd /usr/share/openclash/openclash_chnroute.sh
Thu May 15 03:00:27 2025 cron.err crond[22135]: crond (busybox 1.36.1) started, log level 5
Thu May 15 03:59:59 2025 cron.err crond[3880]: crond (busybox 1.36.1) started, log level 5
Thu May 15 04:59:33 2025 cron.err crond[18098]: crond (busybox 1.36.1) started, log level 5
Thu May 15 05:14:20 2025 daemon.err netdata[9064]: read failed: end of file
Thu May 15 05:14:20 2025 daemon.info netdata[9064]: PARSER ended
Thu May 15 05:14:20 2025 daemon.err netdata[9064]: '/usr/lib/netdata/plugins.d/charts.d.plugin' (pid 29728) disconnected after 14322 successful data collections (ENDs).
Thu May 15 05:14:22 2025 daemon.info netdata[9064]: connected to '/usr/lib/netdata/plugins.d/charts.d.plugin' running on pid 22029
Thu May 15 05:14:23 2025 daemon.info netdata[9064]: RRDSET: chart name 'sensors.temp_thermal_zone0_thermal_thermal_zone0' on host 'LEDE' already exists.
Thu May 15 05:14:23 2025 daemon.info netdata[9064]: RRDSET: chart name 'netdata.plugin_chartsd_sensors' on host 'LEDE' already exists.
Thu May 15 05:59:05 2025 cron.err crond[32358]: crond (busybox 1.36.1) started, log level 5
Thu May 15 06:58:42 2025 cron.err crond[14203]: crond (busybox 1.36.1) started, log level 5
Thu May 15 07:58:14 2025 cron.err crond[28345]: crond (busybox 1.36.1) started, log level 5
Thu May 15 08:03:10 2025 daemon.info netdata[9064]: Initializing file /var/cache/netdata/disk_qops.mmcblk0/main.db.
Thu May 15 08:03:10 2025 daemon.info netdata[9064]: Initializing file /var/cache/netdata/disk_qops.mmcblk0/operations.db.
Thu May 15 08:58:20 2025 cron.err crond[10187]: crond (busybox 1.36.1) started, log level 5
Thu May 15 09:14:26 2025 daemon.err netdata[9064]: read failed: end of file
Thu May 15 09:14:26 2025 daemon.info netdata[9064]: PARSER ended
Thu May 15 09:14:26 2025 daemon.err netdata[9064]: '/usr/lib/netdata/plugins.d/charts.d.plugin' (pid 22029) disconnected after 14322 successful data collections (ENDs).
Thu May 15 09:14:28 2025 daemon.info netdata[9064]: connected to '/usr/lib/netdata/plugins.d/charts.d.plugin' running on pid 14469
Thu May 15 09:14:28 2025 daemon.info netdata[9064]: RRDSET: chart name 'sensors.temp_thermal_zone0_thermal_thermal_zone0' on host 'LEDE' already exists.
Thu May 15 09:14:29 2025 daemon.info netdata[9064]: RRDSET: chart name 'netdata.plugin_chartsd_sensors' on host 'LEDE' already exists.
Thu May 15 09:57:57 2025 cron.err crond[24567]: crond (busybox 1.36.1) started, log level 5
Thu May 15 10:57:30 2025 cron.err crond[6301]: crond (busybox 1.36.1) started, log level 5
Thu May 15 11:57:03 2025 cron.err crond[20469]: crond (busybox 1.36.1) started, log level 5
Thu May 15 12:18:43 2025 cron.err crond[2848]: crond (busybox 1.36.1) started, log level 5
Thu May 15 12:30:58 2025 cron.err crond[27390]: crond (busybox 1.36.1) started, log level 5
Thu May 15 12:33:37 2025 daemon.err uhttpd[1902]: head: standard output: I/O error
Thu May 15 12:46:51 2025 authpriv.info dropbear[23288]: Child connection from 192.168.1.50:6068
Thu May 15 12:46:53 2025 authpriv.warn dropbear[23288]: Login attempt for nonexistent user
Thu May 15 12:47:05 2025 authpriv.info dropbear[23288]: Exit before auth from <192.168.1.50:6068>: Max auth tries reached - user 'is invalid'
Thu May 15 12:47:28 2025 authpriv.info dropbear[23491]: Child connection from 192.168.1.50:6058
Thu May 15 12:47:31 2025 authpriv.notice dropbear[23491]: Password auth succeeded for 'root' from 192.168.1.50:6058
Thu May 15 12:48:16 2025 daemon.err uhttpd[1902]: head: standard output: I/O error
Thu May 15 12:52:32 2025 daemon.info netdata[9064]: RRD database for chart 'sensors.temp_thermal_zone0_thermal_thermal_zone0' on host 'LEDE' is 156.02532 secs in the past (counter #78491, update #78039). Adjusting it to current time.
Thu May 15 12:52:32 2025 daemon.info netdata[9064]: RRD database for chart 'netdata.plugin_chartsd_sensors' on host 'LEDE' is 156.02592 secs in the past (counter #78491, update #78039). Adjusting it to current time.
```

bigbugcc, 2025-05-16:
建议回退版本，新版本问题很多。

galendai, 2025-05-19:
尝试回退2025/4/30的build依旧又这个问题，有可能是OpenClash客户端更新后的问题(0.46.079 Alpha)，在另外一台服务器上独立部署docker的外部Zashboard控制面板了。

Close this issue for now. 感谢帮助 @bigbugcc