bigbugcc/OpenWrts #61: wan口不定时关闭，获取ip失败

adfnkjasd, 2025-05-15:
### 问题描述

wan口绑定在eth0，目前会无规律的不定时down，抓取日志也没有报错
有NIC link is down，NIC是用的转接网口，更换后问题依旧，网线也换过，没有解决
问题是出现在2点55分，5点37分开始是我登录后台查看问题，解决措施是重启wan口

### 重复 issue

- [x] 没有类似的 issue

### 浏览器搜索

- [x] 未找到类似的问题

### 设备型号

x86_64

### 固件版本

openwrt-x86-64-generic-ext4-combined-efi.img.gz

### 详细日志

Thu May 15 02:55:21 2025 daemon.notice netifd: Network device 'eth0' link is down
Thu May 15 02:55:21 2025 daemon.notice netifd: Interface 'wan' has link connectivity loss
Thu May 15 02:55:21 2025 kern.info kernel: [255277.205883] igb 0000:02:00.0 eth0: igb: eth0 NIC Link is Down
Thu May 15 02:56:15 2025 kern.info kernel: [255331.095893] igb 0000:02:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Thu May 15 02:56:15 2025 daemon.notice netifd: Network device 'eth0' link is up
Thu May 15 02:56:15 2025 daemon.notice netifd: Interface 'wan' has link connectivity
Thu May 15 02:56:16 2025 authpriv.info dropbear[4017]: Early exit: Terminated by signal
Thu May 15 02:56:16 2025 authpriv.info dropbear[3861]: Not backgrounding
Thu May 15 05:37:35 2025 daemon.notice netifd: Interface 'wan_6' is now down
Thu May 15 05:37:35 2025 daemon.notice netifd: Network alias '' link is down
Thu May 15 05:37:35 2025 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Thu May 15 05:37:35 2025 daemon.notice netifd: Interface 'wan_6' is disabled
Thu May 15 05:37:35 2025 daemon.notice ttyd[7831]: [2025/05/15 05:37:35:3169] N: rops_handle_POLLIN_netlink: DELADDR
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Withdrawing address record for 2408:8214:204:9bb0:59b4:9b12:1751:7d3a on pppoe-wan.
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address 2408:8214:204:9bb0:59b4:9b12:1751:7d3a.
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Joining mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::59b4:9b12:1751:7d3a.
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Registering new address record for fe80::59b4:9b12:1751:7d3a on pppoe-wan.*.
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Registering new address record for fe80::223:24ff:fe92:1a13 on br-lan.*.
Thu May 15 05:37:35 2025 daemon.info pppd[2953]: Terminating on signal 15
Thu May 15 05:37:35 2025 daemon.info pppd[2953]: Connect time 4416.7 minutes.
Thu May 15 05:37:35 2025 daemon.info pppd[2953]: Sent 793293085 bytes, received 2497466391 bytes.
Thu May 15 05:37:35 2025 daemon.notice ttyd[7831]: [2025/05/15 05:37:35:3229] N: rops_handle_POLLIN_netlink: DELADDR
Thu May 15 05:37:35 2025 daemon.notice ttyd[7831]: [2025/05/15 05:37:35:3229] N: rops_handle_POLLIN_netlink: DELADDR
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Withdrawing address record for 112.232.46.236 on pppoe-wan.
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Leaving mDNS multicast group on interface pppoe-wan.IPv4 with address 112.232.46.236.
Thu May 15 05:37:35 2025 daemon.notice netifd: Network device 'pppoe-wan' link is down
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Interface pppoe-wan.IPv4 no longer relevant for mDNS.
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Withdrawing address record for fe80::59b4:9b12:1751:7d3a on pppoe-wan.
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::59b4:9b12:1751:7d3a.
Thu May 15 05:37:35 2025 daemon.info avahi-daemon[3020]: Interface pppoe-wan.IPv6 no longer relevant for mDNS.
Thu May 15 05:37:36 2025 authpriv.info dropbear[3861]: Early exit: Terminated by signal
Thu May 15 05:37:36 2025 authpriv.info dropbear[6295]: Early exit: Terminated by signal
Thu May 15 05:37:36 2025 authpriv.info dropbear[27261]: Not backgrounding
Thu May 15 05:37:40 2025 daemon.notice netifd: Interface 'wan' is now down
Thu May 15 05:37:40 2025 daemon.info avahi-daemon[3020]: Interface eth0.IPv6 no longer relevant for mDNS.
Thu May 15 05:37:40 2025 daemon.info avahi-daemon[3020]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::231:11ff:fe06:1b0d.
Thu May 15 05:37:40 2025 daemon.info avahi-daemon[3020]: Withdrawing address record for fe80::231:11ff:fe06:1b0d on eth0.
Thu May 15 05:37:40 2025 daemon.notice netifd: Interface 'wan' is disabled
Thu May 15 05:37:40 2025 daemon.notice ttyd[7831]: [2025/05/15 05:37:40:3392] N: rops_handle_POLLIN_netlink: DELADDR
Thu May 15 05:37:40 2025 daemon.notice netifd: Interface 'wan' is enabled
Thu May 15 05:37:40 2025 daemon.notice netifd: Interface 'wan' is setting up now
Thu May 15 05:37:40 2025 daemon.notice netifd: Network device 'eth0' link is down
Thu May 15 05:37:40 2025 daemon.notice netifd: Interface 'wan' has link connectivity loss
Thu May 15 05:37:40 2025 kern.info kernel: [265016.011082] 8021q: adding VLAN 0 to HW filter on device eth0
Thu May 15 05:37:40 2025 daemon.info pppd[27414]: Plugin pppoe.so loaded.
Thu May 15 05:37:40 2025 daemon.info pppd[27414]: PPPoE plugin from pppd 2.4.9
[... 43 lines cut ...]
Thu May 15 05:37:48 2025 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Thu May 15 05:37:48 2025 user.notice network: passwall2: restart when wan ifup
Thu May 15 05:37:48 2025 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Thu May 15 05:37:49 2025 user.notice ddns-scripts[28081]: Ali: PID '28081' started at 2025-05-15 05:37
Thu May 15 05:37:49 2025 user.warn ddns-scripts[28081]: Ali: Service section disabled! - TERMINATE
Thu May 15 05:37:49 2025 user.warn ddns-scripts[28081]: Ali: PID '28081' exit WITH ERROR '1' at 2025-05-15 05:37
Thu May 15 05:37:49 2025 cron.err crond[28933]: crond (busybox 1.36.0) started, log level 8
Thu May 15 05:37:49 2025 authpriv.warn dropbear[29136]: Failed listening on '22': Error listening: Invalid argument
Thu May 15 05:37:49 2025 authpriv.info dropbear[29136]: Not backgrounding
Thu May 15 05:37:51 2025 daemon.info avahi-daemon[3020]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::1c56:8195:4496:82c1.
Thu May 15 05:37:51 2025 daemon.info avahi-daemon[3020]: Joining mDNS multicast group on interface pppoe-wan.IPv6 with address 2408:8215:204:b6a4:1c56:8195:4496:82c1.
Thu May 15 05:37:51 2025 daemon.info avahi-daemon[3020]: Registering new address record for 2408:8215:204:b6a4:1c56:8195:4496:82c1 on pppoe-wan.*.
Thu May 15 05:37:51 2025 daemon.info avahi-daemon[3020]: Withdrawing address record for fe80::1c56:8195:4496:82c1 on pppoe-wan.
Thu May 15 05:37:51 2025 daemon.info avahi-daemon[3020]: Registering new address record for 2408:8214:24a:a8b0::1 on br-lan.*.
Thu May 15 05:37:51 2025 daemon.info avahi-daemon[3020]: Withdrawing address record for 2408:8214:248:98f0::1 on br-lan.
Thu May 15 05:37:51 2025 daemon.info avahi-daemon[3020]: Withdrawing address record for fe80::223:24ff:fe92:1a13 on br-lan.
Thu May 15 05:37:51 2025 daemon.notice netifd: Interface 'wan_6' is now up
Thu May 15 05:37:51 2025 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Thu May 15 05:37:51 2025 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan_6 (pppoe-wan)
Thu May 15 05:37:51 2025 daemon.info dnsmasq[29888]: started, version 2.88 cache disabled
Thu May 15 05:37:51 2025 daemon.info dnsmasq[29888]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset no-auth no-cryptohash no-DNSSEC loop-detect inotify dumpfile
Thu May 15 05:37:51 2025 daemon.info dnsmasq[29888]: using nameserver 202.102.128.68#53
Thu May 15 05:37:51 2025 daemon.info dnsmasq[29888]: using nameserver 202.102.134.68#53
Thu May 15 05:37:51 2025 daemon.info dnsmasq[29888]: read /etc/hosts - 6 names
Thu May 15 05:37:51 2025 user.notice network: passwall2: restart when wan_6 ifup

bigbugcc, 2025-05-16:
I225的网卡？

adfnkjasd, 2025-05-16:
> I225的网卡？

 不，I210，mini PCIe转接出来的

adfnkjasd, 2025-05-16:
> > I225的网卡？
> 
> 不，I210，mini PCIe转接出来的

我目前的解决措施是加了个Watchcat的插件，ping超时3次就重启wan口，但终归不是解决办法

bigbugcc, 2025-05-16:
日志没看到什么有用的信息，更新最新的固件出的问题？新版本硬件兼容好像都有问题，先回退老版本等大雕他们迭代稳定在用吧。

adfnkjasd, 2025-05-16:
> 日志没看到什么有用的信息，更新最新的固件出的问题？新版本硬件兼容好像都有问题，先回退老版本等大雕他们迭代稳定在用吧。

不是最新的，哪个版本忘了但不是最新的，至少是24年的，那先这样吧，等着稳定了更新看看

closed 2025-10-08 by bigbugcc